kamilsss655/uv-k5-firmware-custom #225: [BUG] Please use a title that makes sense

harifhaqim, 2025-05-27:
**Describe the bug**
Unable to type in serial for SMS,even SMS: format doesnt work

**To Reproduce**
Steps to reproduce the behavior:
1. Open serial command with modified makefile command

**Expected behavior**
Should've worked with output

this is my makefile

`ENABLE_CLANG                  := 0
ENABLE_SWD                    := 0
ENABLE_OVERLAY                := 0
ENABLE_LTO                    := 1

# ---- STOCK QUANSHENG FERATURES ----
ENABLE_UART                   := 1
ENABLE_AIRCOPY                := 0
ENABLE_FMRADIO                := 1
ENABLE_NOAA                   := 0
ENABLE_VOICE                  := 0
ENABLE_VOX                    := 1
ENABLE_ALARM                  := 0
ENABLE_TX1750                 := 0
ENABLE_PWRON_PASSWORD         := 1
ENABLE_DTMF_CALLING           := 0

# ---- CUSTOM MODS ----
ENABLE_BIG_FREQ                         := 1
ENABLE_SMALL_BOLD                       := 1
ENABLE_KEEP_MEM_NAME                    := 1
ENABLE_WIDE_RX                          := 1
ENABLE_TX_WHEN_AM                       := 0
ENABLE_F_CAL_MENU                       := 0
ENABLE_SHOW_CHARGE_LEVEL                := 0
ENABLE_REVERSE_BAT_SYMBOL               := 0
ENABLE_NO_CODE_SCAN_TIMEOUT             := 1
ENABLE_SQUELCH_MORE_SENSITIVE           := 0
ENABLE_FASTER_CHANNEL_SCAN              := 1
ENABLE_RSSI_BAR                         := 1
ENABLE_AUDIO_BAR                        := 1
ENABLE_COPY_CHAN_TO_VFO                 := 1
ENABLE_SPECTRUM                         := 1
ENABLE_REDUCE_LOW_POWER                 := 1
ENABLE_BYP_RAW_DEMODULATORS             := 0
ENABLE_BLMIN_TMP_OFF                    := 0
ENABLE_SCAN_RANGES                      := 1
ENABLE_SPECTRUM_COPY_VFO                := 1
ENABLE_SPECTRUM_SHOW_CHANNEL_NAME       := 1
ENABLE_SPECTRUM_CHANNEL_SCAN            := 1
ENABLE_MESSENGER                        := 1
ENABLE_MESSENGER_DELIVERY_NOTIFICATION  := 1
ENABLE_MESSENGER_FSK_MUTE               := 1
ENABLE_MESSENGER_NOTIFICATION           := 1
ENABLE_MESSENGER_UART                   := 1
ENABLE_ENCRYPTION                       := 1`

kamilsss655, 2025-05-30:
This is not a valid bug report, but a code review request since you've changed the makefile.

closed 2025-05-30 by kamilsss655

harifhaqim, 2025-05-31:
but how do I send the SMS: command properly on Windows tho? I cant find the sources online